coolsnowwolf/packages #352: dockerd 和 luci-app-docker 有些文件都有导致安装失败

zhangguanzhang, 2022-01-07:
dockerd 的 Makefile里有文件 `/etc/config/dockerd` `/etc/docker/daemon.json` `/etc/init.d/dockerd`
https://github.com/coolsnowwolf/packages/blob/d6af0a64cf83e5094a88456769a36a2e0ede732e/utils/dockerd/Makefile#L116-L131

但是 `luci-app-docker` 依赖 dockerd，它也有文件这三个文件
```
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-docker/root/etc/config/dockerd
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-docker/root/etc/init.d/dockerd
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-docker/root/etc/docker/daemon.json
```
会导致在安装的时候报错这三个文件冲突：
```
ConfigCollected errors:
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/config/dockerd
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/docker/daemon.json
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/dockerd
	But that file is already provided by package  * dockerd
 * opkg_install_cmd: Cannot install package luci-app-docker.
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/config/dockerd
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/docker/daemon.json
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/dockerd
	But that file is already provided by package  * dockerd
```
@coolsnowwolf PATL

zhangguanzhang, 2022-01-10:
@coolsnowwolf @AmadeusGhost

Beginner-Go, 2022-01-14:
[用这个呢?](https://github.com/lisaac/luci-app-dockerman)

zhangguanzhang, 2022-06-10:
我对比了 openwrt的仓库和编译出的固件，这三个文件都应该是 dockerd 包的，luci-app-docker这个包官方也没有，这个包不应该强这三个文件

coolsnowwolf, 2026-09-05:
已核对当前 master：dockerd 只安装 /etc/init.d/dockerd 和 /etc/config/dockerd，不再安装 issue 所列的 /etc/docker/daemon.json；当前 luci-app-docker 使用 /etc/init.d/docker 与 /etc/config/docker。原三文件重复归属问题已消失，关闭。

closed 2026-09-05 by coolsnowwolf