coolsnowwolf/packages #360: frr升级到8.1.0后，使用lean lede源码编译ACRH17失败

JayXus, 2022-02-19:
frr升级到8.1.0后，使用lean lede源码编译ACRH17失败

checking for LIBYANG (libyang >= 2.0.0)... no
configure: error: libyang (>= 2.0.0) was not found on your system.
make[3]: *** [Makefile:248: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/frr-1794094a65b904ae3a10c51f021c3b7800a9b116/.configured_a950d0882198878b162a57b001ee2d4c] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/frr'
time: package/feeds/packages/frr/compile#20.17#3.42#23.15
    ERROR: package/feeds/packages/frr failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/frr/compile] Error 1

[logs_22.zip](https://github.com/coolsnowwolf/packages/files/8102351/logs_22.zip)

Beginner-Go, 2022-02-19:
别急 我看看

Beginner-Go, 2022-02-19:
更新feeds 应该修好了

JayXus, 2022-02-20:
> 更新feeds 应该修好了

感谢，已经编译通过。

closed 2022-02-20 by JayXus