coolsnowwolf/packages #401: open-vm-tools-12.0.0-19345655 error

naoki66, 2022-04-11:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```
make[3]: Entering directory '/home/alex/lede/feeds/packages/utils/open-vm-tools'
mkdir -p /home/alex/lede/dl
SHELL= flock /home/alex/lede/tmp/.open-vm-tools-12.0.0-19345655.tar.gz.flock -c '       /home/alex/lede/scripts/download.pl "/home/alex/lede/dl" "open-vm-tools-12.0.0-19345655.tar.gz" "ea3755d80ba0e775" "" "https://github.com/vmware/open-vm-tools/releases/download/stable-12.0.0"    '
rm -f /home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/.built
touch /home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/alex/lede/build_dir/target-x86_64_musl/open-345655 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home//libintl-stub/include  -I/home/alex/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/alex/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/alex/linclude -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -for=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655=open-vm-tools-12.0.0-19345655 -Wformat -Wer_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/includ4_gcc-8.4.0_musl/usr/include -I/home/alex/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/alex/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/home/alexiconv-stub/include -I/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/alex/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/alesl/lib -znow -zrelro -L/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/h/lib/libintl-stub/lib -Wl,-rpath-link=/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -liconv " make  -C /home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-1cc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/alex/led0.0-19345655=open-vm-tools-12.0.0-19345655 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/alex/lede/staging_dir/target-x86_64_musl/usaging_dir/target-x86_64_musl/usr/lib/libintl-stub/include" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linuxANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-li ARCH="x86_64" ;
make[4]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655'
Making all in lib
make[5]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib'
Making all in guestRpc
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/guestRpc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/guestRpc'
Making all in auth
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/auth'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/auth'
Making all in backdoor
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/backdoor'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/backdoor'
Making all in asyncsocket
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/asyncsocket'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/asyncsocket'
Making all in sslDirect
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/sslDirect'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/sslDirect'
Making all in pollGtk
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/pollGtk'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/pollGtk'
Making all in poll
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/poll'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/poll'
Making all in dataMap
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dataMap'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dataMap'
Making all in hashMap
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/hashMap'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/hashMap'
Making all in dict
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dict'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dict'
Making all in dynxdr
make[6]: Entering directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dynxdr'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/lib/dynxdr'
[... 213 lines cut ...]
 #define N_(String) (String)

In file included from /home/alex/lede/staging_dir/target-x86_64_musl/usr/include/glib-2.0/glib/gi18n.h:23,
                 from cmdLine.c:34:
/home/alex/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include/libintl.h:42: note: this is the location of the previous definition
 #define N_(String) String

cc1: all warnings being treated as errors
make[7]: *** [Makefile:595: vmtoolsd-cmdLine.o] Error 1
make[7]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/services/vmtoolsd'
make[6]: *** [Makefile:470: all] Error 2
make[6]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/services/vmtoolsd'
make[5]: *** [Makefile:499: all-recursive] Error 1
make[5]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655/services'
make[4]: *** [Makefile:563: all-recursive] Error 1
make[4]: Leaving directory '/home/alex/lede/build_dir/target-x86_64_musl/open-vm-tools-12.0.0-19345655'
[... cut ...]
make[3]: Leaving directory '/home/alex/lede/feeds/packages/utils/open-vm-tools'
time: package/feeds/packages/open-vm-tools/compile#0.97#0.26#1.40
    ERROR: package/feeds/packages/open-vm-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/open-vm-tools/compile] Error 1
make[2]: Leaving directory '/home/alex/lede'
make[1]: *** [package/Makefile:110: /home/alex/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/alex/lede'
make: *** [/home/alex/lede/include/toplevel.mk:230: world] Error 2

Beginner-Go, 2022-04-11:
我看看

naoki66, 2022-04-11:
> 我看看

谢谢大佬，抽空也更新下https://github.com/zerotier/ZeroTierOne的1.8.7哈，直接该版本也是报错。估计需要打patch

Beginner-Go, 2022-04-11:
open-vm-tools先回退 11.3.0

closed 2022-04-11 by Beginner-Go