coolsnowwolf/packages #502: pgyvpn coredump issue

GreatMichaelLee, 2022-08-25:
# Issue template (remove lines from top till here)

Maintainer: @aiamadeus
Environment: (put here arch, model, OpenWrt version)
X86  model @ OpenWrt R22.8.2 / LuCI Master (git-22.236.14286-082687d) Intel N5105@PVE

Description:
pgyvpn cause coredump timely everytime it shows 'disconnected'.

Formating code blocks by wrapping them with pairs of ```
Thu Aug 25 16:40:41 2022 user.notice vpnprogress: disconnected
Thu Aug 25 16:40:42 2022 kern.info kernel: [ 3068.181404] pgyvpnsvr[13356]: segfault at 8 ip 000000000046e53a sp 00007fa4aca84a50 error 4 in pgyvpnsvr[409000+d6000]
Thu Aug 25 16:40:42 2022 kern.info kernel: [ 3068.182817] Code: e8 ae fe ff ff 89 df 85 c0 75 0a e8 40 c0 f9 ff 40 b5 01 eb 05 e8 36 c0 f9 ff 5b 89 e8 5d 41 5c 41 5d 41 5e c3 90 48 83 ec 18 <8b> 7f 08 48 63 d2 e8 7b b5 f9 ff 85 c0 79 14 bf ff 7f 4f 00 48 89

Thu Aug 25 17:24:18 2022 user.notice vpnprogress: disconnected
Thu Aug 25 17:24:18 2022 kern.info kernel: [ 5684.216874] pgyvpnsvr[18567]: segfault at 8 ip 000000000046e53a sp 00007f4eaa786a50 error 4 in pgyvpnsvr[409000+d6000]
Thu Aug 25 17:24:18 2022 kern.info kernel: [ 5684.218645] Code: e8 ae fe ff ff 89 df 85 c0 75 0a e8 40 c0 f9 ff 40 b5 01 eb 05 e8 36 c0 f9 ff 5b 89 e8 5d 41 5c 41 5d 41 5e c3 90 48 83 ec 18 <8b> 7f 08 48 63 d2 e8 7b b5 f9 ff 85 c0 79 14 bf ff 7f 4f 00 48 89

Thu Aug 25 19:18:05 2022 user.notice vpnprogress: disconnected
Thu Aug 25 19:18:05 2022 kern.info kernel: [12510.844056] pgyvpnsvr[8799]: segfault at 8 ip 000000000046e53a sp 00007fe0a166ea50 error 4 in pgyvpnsvr[409000+d6000]
Thu Aug 25 19:18:05 2022 kern.info kernel: [12510.845532] Code: e8 ae fe ff ff 89 df 85 c0 75 0a e8 40 c0 f9 ff 40 b5 01 eb 05 e8 36 c0 f9 ff 5b 89 e8 5d 41 5c 41 5d 41 5e c3 90 48 83 ec 18 <8b> 7f 08 48 63 d2 e8 7b b5 f9 ff 85 c0 79 14 bf ff 7f 4f 00 48 89
Thu Aug 25 19:18:10 2022 user.notice vpnprogress: tryconnect
Thu Aug 25 19:18:10 2022 user.notice vpnprogress: tryconnect
Thu Aug 25 19:18:10 2022 user.notice vpnprogress: login_err
Thu Aug 25 19:18:10 2022 user.notice errcode: 1

```

kbkman, 2022-08-25:
/etc/init.d/pgyvpn restart 后直接就崩溃了吗？

GreatMichaelLee, 2022-08-25:
刚试了，restart不会，应该是vpn不定期会自己disconnect(我也不知道为啥），猜测在disconnect的时候没有优雅的释放一些系统资源（比如socket句柄）导致程序崩溃，鉴于是disconnect的时候发生的崩溃，所以还没有遇到断联的情况，应该后来能connect起来，impact目前没发现影响vpn正常使用。但终归有core不太好。
/etc/init.d/pgyvpn restart  日志，没有core.

Thu Aug 25 21:45:37 2022 user.notice pgyvpn: pgyvpn app filename is /usr/sbin/pgyvpnsvr
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: username is *********
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: log path is /tmp/oray/pgyvpnsvr/log
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: log mask is 7
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: local ip is 192.168.2.1
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: local mask is 255.255.255.0
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: script for p2p init is /usr/share/pgyvpn/p2pinit.sh
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: script for vnc init is /usr/share/pgyvpn/vncinit.sh
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: script for progress is /usr/share/pgyvpn/vpnprogress.sh
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: oraysl filename is /usr/sbin/pgyvpn_oraysl
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: oraysl log path is /tmp/oray/pgyvpn_oraysl/log
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: oraysl status filename is /tmp/oray/pgyvpn_oraysl/status
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: oraysl pid filename is /tmp/oray/pgyvpn_oraysl/pid
Thu Aug 25 21:45:37 2022 user.notice pgyvpn: no use encrypt
Thu Aug 25 21:45:38 2022 user.notice vpnprogress: disconnected
Thu Aug 25 21:45:38 2022 user.notice vpnprogress: tryconnect
Thu Aug 25 21:45:38 2022 user.notice vpnprogress: tryconnect
Thu Aug 25 21:45:39 2022 user.notice vpnprogress: connected

GreatMichaelLee, 2022-08-25:
另外makefile里没有用-g编译选项, 所以有ip我也没法用addr2line显示代码core在哪里。

GreatMichaelLee, 2022-08-25:
> -g 是指？我不太明白（

哦，没事了，我以为是开源拉代码编译的，看样子就是拉了个binary+一堆脚本。那没办法了。

closed 2022-08-26 by Beginner-Go