coolsnowwolf/packages #575: package/feeds/packages/syncthing failed to build with Golang 1.19

dieterplex, 2022-11-04:
Maintainer: @Sunshine-Planet (find it by checking history of the package Makefile)
Environment: x86_64 R22.11.11

Description:
> The version of quic-go you're using can't be built on Go 1.19 yet.

Reverting commit 4bdfbb55 to use golang 1.18 works for me.

Log:

```
Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../vendor/github.com/lucas-clemente/quic-go/internal/qtls/go119.go:6:13: cannot use "The version of quic-go you're using can't be built on Go 1.19 yet. For more details, please see https://github.com/lucas-cle
mente/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.19 yet. F...) as int value in variable declaration

make[3]: *** [Makefile:68: /home/dieter/source/lede/build_dir/target-x86_64_musl/syncthing-1.20.1/syncthing/.built] Error 2
make[3]: Leaving directory '/home/dieter/source/lede/feeds/packages/utils/syncthing'
time: package/feeds/packages/syncthing/compile#2.02#0.98#1.54
    ERROR: package/feeds/packages/syncthing failed to build.
```

Sunshine-Planet, 2022-11-04:
I'm not the maintainer of utils/syncthing, can't help you check this issue.
By the way, where do you get this ID? This bothers me.

dieterplex, 2022-11-04:
Sorry about that I don't verify the detail carefully.
By `git log utils/syncthing` and fount bacc2825 which is a initial commits include a lot of changes in this repo. It looks like the commit history is squashed and bring your name there.

commit da1211e4 (2022-11-06): syncthing: update to 1.22.1

Fixes: #575
closed 2022-11-06 by aiamadeus