coolsnowwolf/packages #581: 更新源码后缺依赖

mitsukileung, 2022-11-09:
WARNING: Makefile 'package/feeds/packages/libesmtp/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/midisport-firmware/Makefile' has a dependency on 'fxload', which does not exist
WARNING: Makefile 'package/feeds/packages/openvswitch/Makefile' has a build dependency on 'python-six/host', which does not exist
WARNING: Makefile 'package/feeds/packages/ovn/Makefile' has a build dependency on 'python-six/host', which does not exist

Beginner-Go, 2022-11-10:
@1582130940 can you look？

1582130940, 2022-11-10:
> @1582130940 can you look？
https://github.com/coolsnowwolf/packages/pull/583
https://github.com/coolsnowwolf/packages/pull/585
https://github.com/coolsnowwolf/lede/pull/10396
我提交了已经，阿巴阿巴

commit a95c0f5c (2022-11-10): libesmtp: Compile with tools/meson (#583)

Fixes: #581
closed 2022-11-10 by aiamadeus