coolsnowwolf/packages #584: 新的packages更新后ERROR

ecrasy, 2022-11-10:
新的packages更新后
重新clone源码
update feeds && install feeds 
出现错误

**LOG：**
Build-Depends: meson/host

Package: libesmtp
Version: 1.1.0-0
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread +libpthread +libopenssl
Conflicts: 
Menu-Depends: 
Provides: 
Section: libs
Category: Libraries
Repository: base
Title: A Library for Posting Electronic Mail
Maintainer: Othmar Truniger <github@truniger.ch>
Source: libesmtp-1.1.0.tar.gz
License: LGPL-2.0-or-later
LicenseFiles: COPYING
Type: ipkg
Description: A Library for Posting Electronic Mail
https://libesmtp.github.io/
Othmar Truniger <github@truniger.ch>
@@

Makefile:26: ../../devel/meson/meson.mk: 没有那个文件或目录
gmake[1]: *** 没有规则可制作目标“../../devel/meson/meson.mk”。 停止。

GreatMichaelLee, 2022-11-10:
同样的问题

ecrasy, 2022-11-10:
刚看代码这个已经被fixed了

closed 2022-11-10 by ecrasy