coolsnowwolf/packages #598: luci-app-music-remote-center 编译还是出错

mitsukileung, 2022-11-12:
configure: error: Unable to link function av_init_packet
configure: WARNING: cache variable pkg_cv_LIBAV_VERSION contains a newline
make[3]: *** [Makefile:80: /home/mitsuki/lede/build_dir/target-x86_64_musl/owntone-28.5/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[3]: Leaving directory '/home/mitsuki/lede/feeds/packages/sound/owntone'
time: package/feeds/packages/owntone/compile#16.50#2.56#20.36
    ERROR: package/feeds/packages/owntone failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/owntone/compile] Error 1
make[2]: Leaving directory '/home/mitsuki/lede'
make[1]: *** [package/Makefile:110: /home/mitsuki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mitsuki/lede'
make: *** [/home/mitsuki/lede/include/toplevel.mk:230：world] 错误 2

Beginner-Go, 2022-11-13:
@1582130940 can you look? 
owntone build error

1582130940, 2022-11-15:
> @1582130940 can you look? owntone build error

Generating index for package ./luci-app-music-remote-center_1-6_all.ipk
Generating index for package ./owntone_28.5-1_x86_64.ipk
我复现不出来啊，建议make distclean重新跑一下

closed 2022-11-15 by mitsukileung