coolsnowwolf/packages #601: conntrack-tools-1.4.7 编译出错

rq1025330, 2022-11-17:
自从conntrack-tools-1.4.6升级到1.4.7就开始报错，报错如下，大佬们这是啥问题呢？

conntrack.c: In function 'nfct_mnl_dump':
conntrack.c:2451:17: warning: implicit declaration of function 'nfct_nlmsg_build_filter'; did you mean 'nfct_nlmsg_build'? [-Wimplicit-function-declaration]
 2451 |                 nfct_nlmsg_build_filter(nlh, filter_dump);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                 nfct_nlmsg_build
conntrack.c:2451:17: warning: nested extern declaration of 'nfct_nlmsg_build_filter' [-Wnested-externs]
conntrack.c: In function 'nfct_stats_cb':
conntrack.c:2531:18: error: 'CTA_STATS_CLASH_RESOLVE' undeclared (first use in this function); did you mean 'CTA_STATS_SEARCH_RESTART'?
 2531 |                 [CTA_STATS_CLASH_RESOLVE] = "clash_resolve",
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
      |                  CTA_STATS_SEARCH_RESTART
conntrack.c:2531:18: note: each undeclared identifier is reported only once for each function it appears in
conntrack.c:2531:18: error: array index in initializer not of integer type
conntrack.c:2531:18: note: (near initialization for 'attr2name')
conntrack.c:2532:18: error: 'CTA_STATS_CHAIN_TOOLONG' undeclared (first use in this function)
 2532 |                 [CTA_STATS_CHAIN_TOOLONG] = "chaintoolong",
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
conntrack.c:2532:18: error: array index in initializer not of integer type
conntrack.c:2532:18: note: (near initialization for 'attr2name')
conntrack.c:2537:39: warning: initialized field overwritten [-Woverride-init]
 2537 |                 [CTA_STATS_MAX + 1] = "unknown1",
      |                                       ^~~~~~~~~~
conntrack.c:2537:39: note: (near initialization for 'attr2name[14]')
conntrack.c:2538:39: warning: initialized field overwritten [-Woverride-init]
 2538 |                 [CTA_STATS_MAX + 2] = "unknown2",
      |                                       ^~~~~~~~~~
conntrack.c:2538:39: note: (near initialization for 'attr2name[15]')
conntrack.c: In function 'do_command_ct':
conntrack.c:3258:51: error: 'NFCT_FILTER_DUMP_STATUS' undeclared (first use in this function); did you mean 'NFCT_FILTER_DUMP_MAX'?
 3258 |                                                   NFCT_FILTER_DUMP_STATUS,
      |                                                   ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                   NFCT_FILTER_DUMP_MAX
make[7]: *** [Makefile:696: conntrack.o] Error 1
make[7]: Leaving directory '/home/rq/lede/build_dir/target-aarch64_cortex-a53_musl/conntrack-tools-1.4.7/src'
make[6]: *** [Makefile:738: all-recursive] Error 1
make[6]: Leaving directory '/home/rq/lede/build_dir/target-aarch64_cortex-a53_musl/conntrack-tools-1.4.7/src'
make[5]: *** [Makefile:515: all] Error 2
make[5]: Leaving directory '/home/rq/lede/build_dir/target-aarch64_cortex-a53_musl/conntrack-tools-1.4.7/src'
make[4]: *** [Makefile:541: all-recursive] Error 1
make[4]: Leaving directory '/home/rq/lede/build_dir/target-aarch64_cortex-a53_musl/conntrack-tools-1.4.7'
make[3]: *** [Makefile:83: /home/rq/lede/build_dir/target-aarch64_cortex-a53_musl/conntrack-tools-1.4.7/.built] Error 2
make[3]: Leaving directory '/home/rq/lede/feeds/packages/net/conntrack-tools'
time: package/feeds/packages/conntrack-tools/compile#25.58#22.99#42.24
    ERROR: package/feeds/packages/conntrack-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/conntrack-tools/compile] Error 1
make[2]: Leaving directory '/home/rq/lede'
make[1]: *** [package/Makefile:110: /home/rq/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rq/lede'
make: *** [/home/rq/lede/include/toplevel.mk:230：world] 错误 2
rq@rq:~/lede$

1582130940, 2022-11-18:
https://github.com/coolsnowwolf/lede/issues/10440
已解决，请更新源码后重试

closed 2022-11-18 by aiamadeus

rq1025330, 2022-11-18:
已经好了