coolsnowwolf/packages #688: Github Action编译x86固件smartdns报错

bi4tms, 2023-05-27:
[image]

picmax, 2023-05-27:
I don't have this issue. Please try to pull the official copy from https://github.com/pymumu/openwrt-smartdns/blob/master/Makefile to see if works.

bi4tms, 2023-05-27:
> I don't have this issue. Please try to pull the official copy from https://github.com/pymumu/openwrt-smartdns/blob/master/Makefile to see if works.

After directly overwrite the file 'Makefile' in package/net/smartdns by the same file from your repository 'package/net/smartdns', someone else can build smartdns successfully. So the problem may come frome the format when codes are copyed and pasted.

picmax, 2023-05-27:
It is possible but I don't have any problem with either file. Perhaps some build environments are more sensitive/picky than others. I will try to resubmit a pull request.

bi4tms, 2023-05-27:
> It is possible but I don't have any problem with either file. Perhaps some build environments are more sensitive/picky than others. I will try to resubmit a pull request.

Thank you for the resubmition.

picmax, 2023-05-27:
Actually, I downloaded the Makefile from both pymumu and lede repositories and compared them. The files are exactly the same. Not sure what caused your problem.

bi4tms, 2023-05-27:
> Actually, I downloaded the Makefile from both pymumu and lede repositories and compared them. The files are exactly the same. Not sure what caused your problem.

I just downloaded the file 'Makefile' from lean's repository and yours and compared two files. SHA1 and MD5 of the two files were different.

picmax, 2023-05-27:
You were right. Tabs were changed into spaces, causing the problem. New pull request submitted.

closed 2023-05-28 by bi4tms