coolsnowwolf/packages #690: open-vm-tools编译报错

bleach1991, 2023-06-19:
autoreconf: Leaving directory `.'
(cd /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/./; if [ -x ./configure ]; then find /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r chmod u+w; find /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r -n1 cp --remove-destination /op/mod/lede-x86_64/scripts/config.guess; find /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r chmod u+w; find /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r -n1 cp --remove-destination /op/mod/lede-x86_64/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CPPFLAGS="-I/op/mod/lede-x86_64/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/op/mod/lede-x86_64/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/op/mod/lede-x86_64/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   " LDFLAGS="-L/op/mod/lede-x86_64/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/op/mod/lede-x86_64/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro   -liconv "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --without-icu --disable-multimon --disable-docs --disable-tests --without-gtkmm --without-gtkmm3 --without-xerces --without-pam --disable-grabbitmqproxy --disable-vgauth --disable-deploypkg --without-root-privileges --without-kernel-modules --without-dnet --with-tirpc --without-x --without-gtk2 --without-gtk3 --without-xerces --enable-resolutionkms=no ; fi; )
configure: WARNING: unrecognized options: --disable-nls, --disable-grabbitmqproxy
configure: loading site script /op/mod/lede-x86_64/include/site/x86_64
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking for x86_64-openwrt-linux-gcc... x86_64-openwrt-linux-musl-gcc
checking whether the C compiler works... no
configure: error: in `/op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:150: /op/mod/lede-x86_64/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/op/mod/lede-x86_64/feeds/packages/utils/open-vm-tools'
time: package/feeds/packages/open-vm-tools/compile#9.39#0.28#10.60
    ERROR: package/feeds/packages/open-vm-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/open-vm-tools/compile] Error 1
make[2]: Leaving directory '/op/mod/lede-x86_64'
make[1]: *** [package/Makefile:110: /op/mod/lede-x86_64/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/op/mod/lede-x86_64'
make: *** [/op/mod/lede-x86_64/include/toplevel.mk:231: world] Error 2

x86_64版本前两天open-vm-tools版本由12.0.0降级到11.3.0后报错

greennyreborn, 2023-06-21:
一样的问题

ywt114, 2023-06-23:
因为官方的open-vm-tools没有回滚11.3，所以只能自己回滚，测试正常
把feeds.conf.default里的浅克隆改成完整克隆
cd lede
sed -i "s/src-git packages/src-git-full packages/g" feeds.conf.default
./scripts/feeds update -a
cd feeds/packages
git reset --hard fa96401fae81b7fd578403ea57ee989f1cf323b3^
cd ../..
./scripts/feeds install -a
自己回滚到之前版本，就正常了

ywt114, 2023-06-23:
git reset --hard 后面的是完整fa96401fae81b7fd578403ea57ee989f1cf323b3^

ywt114, 2023-06-23:
或者在git reset 之前执行git log自己选择回滚记录

xiaotan8, 2023-06-23:
> 

云编译有解决办法吗？

xiaotan8, 2023-06-24:
rm -rf feeds/packages/utils/open-vm-tools
svn co https://github.com/xiaotan8/packages/trunk/utils/open-vm-tools           feeds/packages/utils/open-vm-tools
还好之前拉过12.0的代码。

ywt114, 2023-06-24:
> 云编译有解决办法吗？

我也是云编译，只是把回滚方法贴出来，云编译道理一样，加条项目而已，原理一样。或者svn co覆盖open-vm-tools，都一样效果。

xiaotan8, 2023-06-24:
大佬已撤销该提交，散会。

closed 2023-06-24 by bleach1991