coolsnowwolf/packages #697: Can anyone help to update zerotier to 1.12.1?

GreatMichaelLee, 2023-08-28:
there is a new version of ZT 1.12.1 and RN said there are significant performance improvement, can any one help to upgrade here? I tried directly change make file with updating version# and SHA256, and compile faild, seems need a build patch...

MilesPoupart, 2023-09-01:
maybe refer to https://github.com/mwarning/zerotier-openwrt

GreatMichaelLee, 2023-09-01:
> maybe refer to https://github.com/mwarning/zerotier-openwrt

谢谢，我把你这个下面的patch目录覆盖过来，然后Makefile里DEPENDS 加一个 +libatomic后，在lede上编译过了。建议哪位大佬pull request一下。

avsba001, 2023-09-03:
> > maybe refer to https://github.com/mwarning/zerotier-openwrt
> 
> 谢谢，我把你这个下面的patch目录覆盖过来，然后Makefile里更新版本号/sha256，以及在DEPENDS 加一个 +libatomic后，在lede上编译过了。建议哪位大佬pull request一下。

我直接用官方packages替换lede 编译过的1.12.1

closed 2023-09-14 by aiamadeus