coolsnowwolf/packages #707: phtunnel 无法启动

lxhao61, 2023-09-06:
phtunnel 无法启动，报错提示如下：
phtunnel is not running

尝试使用替换办法（替换 19.07 上运行成功的 phtunnel 程序及配置）无法解决，一样报错。
尝试按照官方提供的 IPK 包在本源码编译好的固件（无 phtunnel 固件）上手工安装；主程序成功安装，luci 报错无法安装。

aiamadeus, 2023-09-14:
kmod-tun 有吗？系统日志呢

lxhao61, 2023-09-14:
> kmod-tun 有吗？系统日志呢

有 kmod-tun。斐讯 K3 系统日志（phtunnel 相关）如下：
.....
Thu Sep 14 21:53:34 2023 user.notice oray: wan interface up and phtunnel restart
Thu Sep 14 21:53:34 2023 daemon.notice procd: /etc/rc.d/S80tailscale: 2023/09/14 13:53:34 logtail: upload: log upload of 528 bytes compressed failed: Post "https://log.tailscale.io/c/tailnode.log.tailscale.io/6044ff1559b2a0e4df2ee6b0e5dae0d2ee648c7ea96e97d12b4eca360734f766": context canceled
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 logtail started
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Program starting: v1.42.0-ERR-BuildInfo, Go 1.20.2: []string{"/usr/sbin/tailscaled", "--port", "41641", "--state", "/etc/tailscale/tailscaled.state"}
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 LogID: 42faea44414fc669266ccb3f064b1921cf19d0eade83f115bfdb4bdfe6c55a21
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 logpolicy: using system state directory "/var/lib/tailscale"
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 wgengine.NewUserspaceEngine(tun "tailscale0") ...
Thu Sep 14 21:53:35 2023 user.err : jail: can't read from child
Thu Sep 14 21:53:35 2023 user.err : jail: can't write to parent
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 logtail: dial "log.tailscale.io:443" failed: dial tcp: lookup log.tailscale.io on [::1]:53: read udp [::1]:33955->[::1]:53: read: connection refused (in 32ms), trying bootstrap...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 trying bootstrapDNS("derp9c.tailscale.com", "155.138.243.219") for "log.tailscale.io" ...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 setting link attributes: netlink receive: no such file or directory
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 router: v6nat = true
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 dns: [rc=unknown ret=direct]
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 dns: using "direct" mode
Thu Sep 14 21:53:35 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 dns: using *dns.directManager
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 link state: interfaces.State{defaultRoute=pppoe-wan ifs={br-lan:[192.168.6.1/24 fd4a:9e1c:fc84::1/60] eth0.3:[10.52.118.246/20] eth0.4:[189.131.122.246/19] pppoe-wan:[218.6.202.243/32]} v4=true v6=true}
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 magicsock: disco key = d:3b2d521bf0410fe4
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Creating WireGuard device...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Bringing WireGuard device up...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Bringing router up...
Thu Sep 14 21:53:35 2023 daemon.notice netifd: Interface 'wan_6' is now up
Thu Sep 14 21:53:35 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/radvd
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Clearing router settings...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Starting network monitor...
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Engine created.
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 pm: migrating "_daemon" profile to new format
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 external route: up
Thu Sep 14 21:53:35 2023 user.notice ucitrack: Setting up /etc/config/wireless reload dependency on /etc/config/network
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 logpolicy: using system state directory "/var/lib/tailscale"
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 got LocalBackend in 324ms
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Start
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Backend: logs: be:42faea44414fc669266ccb3f064b1921cf19d0eade83f115bfdb4bdfe6c55a21 fe:
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 monitor: RTM_NEWROUTE: src=, dst=240e:398:31f:4a1a::/64, gw=, outif=1, table=254
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 monitor: RTM_NEWROUTE: src=, dst=240e:39a:37a:c110::/60, gw=, outif=1, table=254
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 monitor: RTM_NEWROUTE: src=, dst=240e:39a:37a:c110::/64, gw=, outif=7, table=254
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 monitor: RTM_NEWROUTE: src=, dst=, gw=fe80::2a41:c6ff:fefb:71f9, outif=12, table=254
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 monitor: RTM_NEWROUTE: src=, dst=, gw=fe80::2a41:c6ff:fefb:71f9, outif=12, table=254
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 [RATELIMIT] format("monitor: %s: src=%v, dst=%v, gw=%v, outif=%v, table=%v")
Thu Sep 14 21:53:35 2023 daemon.err tailscaled[3153]: 2023/09/14 13:53:35 Switching ipn state NoState -> NeedsLogin (WantRunning=false, nm=false)
[... cut ...]
Thu Sep 14 21:55:53 2023 daemon.err sing-box[8881]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Thu Sep 14 21:55:58 2023 daemon.err sing-box[8889]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Thu Sep 14 21:56:03 2023 daemon.err sing-box[8897]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Thu Sep 14 21:56:03 2023 daemon.info procd: Instance sing-box::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

另外同样源码编译的 Newifi 3 固件的 phtunnel 与 pgyvpn 都可正常启动，猜测应该是没有准确编译对应芯片。

aiamadeus, 2023-09-15:
k3 这个 cpu 是博通，表面上是 armv7 但没有 fpu 支持。。

lxhao61, 2023-09-15:
> k3 这个 cpu 是博通，表面上是 armv7 但没有 fpu 支持。。

是的。希望大神修订一下编译文件可正确识别。

aiamadeus, 2023-09-16:
我更新了一下，feeds 你用最新的，lede 你用 https://github.com/coolsnowwolf/lede/commit/f12c5e7

OpenWrtLi, 2023-09-16:
> 我更新了一下，给你用最新的，lede 你用[coolsnowwolf/lede@ f12c5e7](https://github.com/coolsnowwolf/lede/commit/f12c5e7)

x86 最新6.1内核编译出错

Package pgyvpn is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:133: /home/li/lede6.1/bin/targets/x86/64/packages/pgyvpn_3.1.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/li/lede6.1/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/compile#0.32#0.09#2.25
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/compile] Error 1
make[2]: Leaving directory '/home/li/lede6.1'
make[1]: *** [package/Makefile:110: /home/li/lede6.1/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/li/lede6.1'

aiamadeus, 2023-09-16:
lede 不能用最新的提交，你找我反馈没用。

coolsnowwolf, 2023-09-16:
已经解决

OpenWrtLi, 2023-09-17:
> 已经解决

lean大佬牛逼

lxhao61, 2023-09-17:
> 我更新了一下，feeds 你用最新的，lede 你用 [coolsnowwolf/lede@f12c5e7](https://github.com/coolsnowwolf/lede/commit/f12c5e7)

测试了，phtunnel 在斐讯 K3 上可以正常使用了；但 pgyvpn 在斐讯 K3 上还是无法使用，报错如下（如之前情况）：
[image]

斐讯 K3 系统日志（pgyvpn 相关）如下：
......
Sun Sep 17 12:41:10 2023 daemon.err uhttpd[1999]: cat: can't open '/etc/bench.log': No such file or directory
Sun Sep 17 12:41:44 2023 daemon.err uhttpd[1999]: Command failed: ubus call service delete { "name": "pgyvpn" } (Not found)
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: pgyvpn app filename is /usr/sbin/pgyvpnsvr
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: username is lxhao61
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: log path is /tmp/oray/pgyvpnsvr/log
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: log mask is 7
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: local ip is 192.168.6.1
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: local mask is 255.255.255.0
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: script for p2p init is /usr/share/pgyvpn/p2pinit.sh
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: script for vnc init is /usr/share/pgyvpn/vncinit.sh
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: script for progress is /usr/share/pgyvpn/vpnprogress.sh
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: oraysl filename is /usr/sbin/pgyvpn_oraysl
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: oraysl log path is /tmp/oray/pgyvpn_oraysl/log
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: oraysl status filename is /tmp/oray/pgyvpn_oraysl/status
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: oraysl pid filename is /tmp/oray/pgyvpn_oraysl/pid
Sun Sep 17 12:41:44 2023 user.notice pgyvpn: no use encrypt
Sun Sep 17 12:42:10 2023 daemon.info procd: Instance pgyvpn::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Sep 17 12:42:47 2023 daemon.err uhttpd[1999]: cat: can't open '/etc/bench.log': No such file or directory

另外 Newifi 3 固件的 pgyvpn 可以正常启动，表现一切正常（显示已经成功组网），就是管理平台显示一直离线，从而造成还是无法使用。

hue715, 2024-01-29:
在pgyvpn的`Makefile`文件里增加了下载libssl.so.1.1,libcrypto.so.1.1，并安装到/usr/lib，应该可以临时解决问题
https://github.com/coolsnowwolf/packages/blob/ad79132d2f8084979ee1c1e3210d067b13a3abc5/net/pgyvpn/Makefile#L67-L99
https://github.com/coolsnowwolf/packages/blob/ad79132d2f8084979ee1c1e3210d067b13a3abc5/net/pgyvpn/Makefile#L110-L115

lxhao61, 2024-01-31:
> 在pgyvpn的`Makefile`文件里增加了下载libssl.so.1.1,libcrypto.so.1.1，并安装到/usr/lib，应该可以临时解决问题
>
> https://github.com/coolsnowwolf/packages/blob/ad79132d2f8084979ee1c1e3210d067b13a3abc5/net/pgyvpn/Makefile#L67-L99 [...]

[报错](https://github.com/lxhao61/Actions-Lean-OpenWrt/actions/runs/7716907043/job/21034802557)，无法成功编译。

hue715, 2024-01-31:
> [报错](https://github.com/lxhao61/Actions-Lean-OpenWrt/actions/runs/7716907043/job/21034802557)，无法成功编译。

看来还要改改，对于`arm_cortex-a9`要用`arm_cortex-a9_neon-musl-unknown`
另外还要revert https://github.com/coolsnowwolf/lede/commit/61625a6cf9e10f0f64b40201ad7242661fed91ce

hue715, 2024-01-31:
> [报错](https://github.com/lxhao61/Actions-Lean-OpenWrt/actions/runs/7716907043/job/21034802557)，无法成功编译。

看来oray网上没有与`arm_cortex-a9-musl-unknown`配套的文件

lxhao61, 2024-01-31:
> > [报错](https://github.com/lxhao61/Actions-Lean-OpenWrt/actions/runs/7716907043/job/21034802557)，无法成功编译。
> 
> 看来oray网上没有与`arm_cortex-a9-musl-unknown`配套的文件，那就只能回退到openssl v1.1编译出文件再放入

测试编译的 Newifi 3 固件， pgyvpn 可以正常启动，但故障依旧：表现一切正常（显示已经成功组网），就是管理平台显示一直离线，从而造成还是无法使用。

coolsnowwolf, 2026-09-05:
已在 b14905c87952c576eab3ffd68764ea65459e5337 修复并验证。问题根因是厂商 pgyvpn 3.1.0 仍动态依赖 libssl.so.1.1/libcrypto.so.1.1，而当前系统仅提供 OpenSSL 3；包此前既无法通过依赖检查，运行时也会缺库退出。现在由包源码交叉编译 OpenSSL 1.1.1w 兼容共享库并随 pgyvpn 安装，不影响系统 OpenSSL 3。验证结果：x86_64 使用 -j30 完整编译成功，反解 pgyvpn_3.1.0-2_x86_64.ipk 后两库 SONAME 正确，musl loader 可解析 pgyvpnsvr 全部依赖；另使用仓库现有 arm_cortex-a9 soft-float 工具链 -j30 交叉编译成功，厂商 K3 二进制与兼容库均为 ARMv7 且无 VFP 参数 ABI，动态依赖匹配。原先 K3 下载/编译缺少对应 1.1 库的问题已消除。

closed 2026-09-05 by coolsnowwolf