coolsnowwolf/packages #799: `docker`错误日志刷屏

devioa, 2024-04-27:
`daemon.err dockerd[8884]: 2024/04/27 09:48:32 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)`
系统日志里这样的报错巨多，怀疑是 8f8ca4018a12a91b4b6bc97c25f8c07e8052761e 更新25版本的问题，之前24的版本没出现过
[image]

devioa, 2024-04-28:
我自己fork过后把这个commit revert过后，刷上观察了一天系统日志已经没有这个报错了，可以确认就是25版本的问题

HHJLKK, 2024-04-29:
https://github.com/moby/moby/issues/47448
https://github.com/moby/moby/pull/47715
他这个是dockerd里的moby出问题，或许单独更新一下moby看看能否解决问题？
实在不行回退到24.x也可以

devioa, 2024-04-29:
> [moby/moby#47448](https://github.com/moby/moby/issues/47448) [moby/moby#47715](https://github.com/moby/moby/pull/47715) 他这个是dockerd里的moby出问题，或许单独更新一下moby看看能否解决问题？ 实在不行回退到24.x也可以

应该可行，如果上游已经修复了的话

qingtian110, 2024-05-02:
也可以更新到 [dockerd 26.1.1](https://github.com/moby/moby/releases/tag/v26.1.1)，最新的版本也没有出现错误日志。

[image]
[image]
[image]

devioa, 2024-05-05:
> 也可以更新到 [dockerd 26.1.1](https://github.com/moby/moby/releases/tag/v26.1.1)，最新的版本也没有出现错误日志。 [...]

@HHJLKK 那大佬看看有没有时间更新一下

devioa, 2024-05-07:
刚更新的`docker 26.1.0`这个版本还是会有`http: superfluous response.WriteHeader`错误日志
[image]
[image]

qingtian110, 2024-05-07:
不过令人疑惑的是，#806 都升级 26.1.0 了，为什么不直接升级到 26.1.1 的啊

qingtian110, 2024-05-07:
懂了  copy 的 openwrt 上游的  https://github.com/openwrt/packages/pull/24064

devioa, 2024-05-09:
26.1.1还是有相同的错误日志啊，完全没解决……
[image]
[image]

qingtian110, 2024-05-09:
玄学。。。

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.3, r23809-234f1a2efa
 -----------------------------------------------------
[root@OpenWrt ~]# dockerd --version
Docker version 26.1.1, build ac2de55
[root@OpenWrt ~]# logread | grep dockerd
Thu May  9 04:04:14 2024 user.notice dockerd-init: Bridge device 'docker0' already defined in network config
Thu May  9 04:04:25 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:25.982447711+08:00" level=warning msg="Running experimental build"
Thu May  9 04:04:27 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:27.866153629+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Thu May  9 04:04:27 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:27.868187420+08:00" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Thu May  9 04:04:30 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:30.019830588+08:00" level=warning msg="Error (Unable to complete atomic operation, key modified) deleting object [endpoint af6c40632b5e68f7e2e4d2fee9f73ad01b3ecfaf4a19e885786928ba4a54682f 8d56f344455c2f77b29fbddd0db661fa18786de181b3bb582e4241e38322c66a], retrying...."
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.033356964+08:00" level=warning msg="path in container /dev/dri/card0 already exists in privileged mode" container=6b3212b405e61f5df82f3ad35b7ae494adb4e29f7c79d7905a0f6356f3842a67
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.033472464+08:00" level=warning msg="path in container /dev/dri/card1 already exists in privileged mode" container=6b3212b405e61f5df82f3ad35b7ae494adb4e29f7c79d7905a0f6356f3842a67
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.033509797+08:00" level=warning msg="path in container /dev/dri/renderD128 already exists in privileged mode" container=6b3212b405e61f5df82f3ad35b7ae494adb4e29f7c79d7905a0f6356f3842a67
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.710935672+08:00" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.711176006+08:00" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.711237547+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Thu May  9 04:04:31 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:31.711616422+08:00" level=info msg="loading plugin \"io.containerd.ttrpc.v1.pause\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Thu May  9 04:04:32 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:32.242318631+08:00" level=warning msg="WARNING: bridge-nf-call-iptables is disabled"
Thu May  9 04:04:32 2024 daemon.err dockerd[14089]: time="2024-05-09T04:04:32.242420714+08:00" level=warning msg="WARNING: bridge-nf-call-ip6tables is disabled"
[root@OpenWrt ~]#
```

```
[root@OpenWrt ~]# docker ps
CONTAINER ID   IMAGE             COMMAND            CREATED      STATUS        PORTS                                       NAMES
6b3212b405e6   local/debian:12   "/usr/sbin/init"   6 days ago   Up 14 hours   0.0.0.0:2222->2222/tcp, :::2222->2222/tcp   debian
[root@OpenWrt ~]#
```

devioa, 2024-05-09:
刚刚玩游戏突然断流420了，然后一看日志刷满了……
[image]

qingtian110, 2024-05-09:
OpenWrt 欢迎你  
[image]

closed 2024-07-11 by devioa