coolsnowwolf/packages #812: softethervpn5能否升个级

wangtieying, 2024-05-14:
softethervpn5能否升个级，或者升个5.02以上的版本

625045508, 2024-06-19:
目前 SoftEtherVPN 不稳  3台电脑连接，就会导致全部失联，用老版本的就没问，好奇怪

zhouzhiwen2000, 2024-07-18:
升级到5.02.5180了

625045508, 2024-07-18:
经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳

625045508, 2024-07-18:
> > 经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳
> 
> 但是最新的服务端配置不了

能配置，检查你的443端口，是不是被占用了，导致最新版无法连接

wangtieying, 2024-07-18:
> > > 经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳
> > 
> > 但是最新的服务端配置不了
> 
> 能配置，检查你的443端口，是不是被占用了，导致最新版无法连接

但是我看这会升级的是5185

625045508, 2024-07-18:
> > > > 经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳
> > > 
> > > 
> > > 但是最新的服务端配置不了
> > 
> > 
> > 能配置，检查你的443端口，是不是被占用了，导致最新版无法连接
> 
> 但是我看这会升级的是5185

哦哦哦， 5185 没测过

zhouzhiwen2000, 2024-07-22:
> > > > > 经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳
> > > > 
> > > > 
> > > > 但是最新的服务端配置不了
> > > 
> > > 
> > > 能配置，检查你的443端口，是不是被占用了，导致最新版无法连接
> > 
> > 
> > 但是我看这会升级的是5185
> 
> 哦哦哦， 5185 没测过

写错了，是5185，目前的最新版。

625045508, 2024-07-22:
> > > > > > 经过详细测试，5.02.5180-1 不稳定，莫名其妙断连接，降级到 5.01.9674-1，目前使用最稳
> > > > > 
> > > > > 
> > > > > 但是最新的服务端配置不了
> > > > 
> > > > 
> > > > 能配置，检查你的443端口，是不是被占用了，导致最新版无法连接
> > > 
> > > 
> > > 但是我看这会升级的是5185
> > 
> > 
> > 哦哦哦， 5185 没测过
> 
> 写错了，是5185，目前的最新版。在ipv4和ipv6下测试了几天，目前没有遇到断连问题。

是的，昨天编译了 5.02.5185-1， 一直用到今天，没遇到断连问题

Mike-qian, 2024-11-23:
可以通过软件包升级的时候选择允许覆盖冲突的包文件来更新这个，软件包源可以到该处:[kiddin9](https://github.com/kiddin9/kwrt-packages?tab=readme-ov-file#%E4%BD%BF%E7%94%A8%E8%BD%AF%E4%BB%B6%E6%BA%90%E8%AF%B7%E5%88%A0%E9%99%A4-opkg%E9%85%8D%E7%BD%AEetcopkgconf-%E4%B8%AD%E7%9A%84-option-check_signature)选择适合自己的架构

coolsnowwolf, 2026-09-05:
当前 master 的 net/softethervpn5 已升级到 5.02.5187，高于 issue 请求的 5.02；评论中也已有 5.02.5185 在 IPv4/IPv6 下连续使用正常的反馈。版本请求已完成，关闭。

closed 2026-09-05 by coolsnowwolf