coolsnowwolf/packages #828: 云编译uugamebooster失败

smallprogram, 2024-06-20:
uugamebooster编译失败
自从openwrt telephony提交了如下记录后，uugamebooster就开始编译失败了

SHA | Author | Date | Message
-- | -- | -- | --
b2a5bc8 | Jiri Slachta | Tue Jun 18 22:50:44 2024 +0200 | Merge pull request #871 from dangowrt/asterisk-chan-lantiq-rotary-patches
96db1a2 | Daniel Golle | Tue Jun 18 20:48:42 2024 +0100 | asterisk-chan-lantiq: import patches to improve pulse-dial phones

breakings, 2024-06-22:
目前需要更新到v6.3.10

### ➜ curl -L -s -k -H "Accept:text/plain" "http://router.uu.163.com/api/plugin?type=openwrt-aarch64"   
http://uu.gdl.netease.com/uuplugin/openwrt-aarch64/v6.3.10/uu.tar.gz,94538a4f40dca41f508a0a9cb5958163#                                                      
➜  ~ curl -L -s -k -H "Accept:text/plain" "http://router.uu.163.com/api/plugin?type=openwrt-x86_64" 
http://uu.gdl.netease.com/uuplugin/openwrt-x86_64/v6.3.10/uu.tar.gz,17fb275fe666ec3d03ba3b41c2441156#                                                       
➜  ~ curl -L -s -k -H "Accept:text/plain" "http://router.uu.163.com/api/plugin?type=openwrt-arm"   
http://uu.gdl.netease.com/uuplugin/openwrt-arm/v6.3.10/uu.tar.gz,633596a922256c2a5f8d9373620ea439#                                                          
➜  ~ curl -L -s -k -H "Accept:text/plain" "http://router.uu.163.com/api/plugin?type=openwrt-mipsel"
http://uu.gdl.netease.com/uuplugin/openwrt-mipsel/v6.3.10/uu.tar.gz,022c616f84bd1ed5466eb00193376ad5#

smallprogram, 2024-06-22:
> 目前需要更新到v6.3.10
>
> ➜ curl -L -s -k -H "Accept:text/plain" "http://router.uu.163.com/api/plugin?type=openwrt-aarch64" [...]

你好，是否可以提交一个PR更新一下库中的uugame。

breakings, 2024-06-23:
> 你好，是否可以提交一个PR更新一下库中的uugame。

奇怪，我更新版本和哈希值后下载地址 403，但手动复制链接又可以下载。😹

mitsukileung, 2024-06-23:
> 奇怪，我更新版本和哈希值后下载地址 403，但手动复制链接又可以下载。😹

一样的问题，就算手动改了MD5后还是出错

oXnMe, 2024-06-26:
我在本地测试，再用美国，日本和国内的VPS测试了下，发现是网易加了限制，本地和国内的VPS可以正常请求包，但是美国和日本的就报403

breakings, 2024-06-27:
> ~我在本地测试，再用美国，日本和国内的VPS测试了下，发现是网易加了限制，本地和国内的VPS可以正常请求包，但是美国和日本的就报403~
> 
> emmmm，刚发上来又去试了下，又可以了，我再去编译看看
> 
> Update：还是会报403(图1)，我直接ssh上去试了下curl又没问题(图2)
> 
> ![图1](https://private-user-images.githubusercontent.com/1726458/343012409-95f141f0-c2c7-4574-b5b2-2d76f0e3b589.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NjM1MzcsIm5iZiI6MTcxOTQ2MzIzNywicGF0aCI6Ii8xNzI2NDU4LzM0MzAxMjQwOS05NWYxNDFmMC1jMmM3LTQ1NzQtYjViMi0yZDc2ZjBlM2I1ODkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjdUMDQ0MDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzZmMGFlYjA0ZDhjY2Q3ZDZhY2I5ZDc2N2ViM2RiOWEzNTExMTJmN2RlZjk5Zjc4NWJjNzhjZTkyZGVjOGJiYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yaxHJEPIw1-SvS_Bxk19fRirS9OURIAgNuIfty15Rro)
> 
> ![图2](https://private-user-images.githubusercontent.com/1726458/343012065-0ccf5f44-7415-480d-befa-e8456cfce5d6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NjM1MzcsIm5iZiI6MTcxOTQ2MzIzNywicGF0aCI6Ii8xNzI2NDU4LzM0MzAxMjA2NS0wY2NmNWY0NC03NDE1LTQ4MGQtYmVmYS1lODQ1NmNmY2U1ZDYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjdUMDQ0MDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDJkY2JjZTlkZDU5YWQ0YWFhOTNmNzU1ZWYzODQyZTRjMDMzYjNjZTY4ODk5MzdmMmQwMWNkYzAwMTYzZjY4YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.trg3xIm74Hyh2a6-6uhyIRcOzecoMKMH8ZcnvqecUNw)
> 
> Update 1：测试的时候发现(图3)，然后注意到UU这个包是放在阿里云OSS的，不确定以前是不是在OSS的，怀疑有没有可能是`PKG_SOURCE_URL`里的`?`导致请求OSS的鉴权信息丢失
> 
> ![图3](https://private-user-images.githubusercontent.com/1726458/343082374-feab05d7-c597-49b7-a916-a7a8d5ee5800.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NjM1MzcsIm5iZiI6MTcxOTQ2MzIzNywicGF0aCI6Ii8xNzI2NDU4LzM0MzA4MjM3NC1mZWFiMDVkNy1jNTk3LTQ5YjctYTkxNi1hN2E4ZDVlZTU4MDAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjdUMDQ0MDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjlhNmM2MzlhNmFlNTBkMTNhMGNlNWZhYWE5NTEyYmY3NWY2ZGM0ZDcxOWUzOWNjODk2YjZhNGZmNDYwYjk5NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.mY_WUsukUswCuF8vV9TD76CyQj8QS1dWnSBJ8K4GwHc)
> 
> 尝试修改Makefile：
> 
> ![图4](https://private-user-images.githubusercontent.com/1726458/343082976-c798168b-62bb-4d02-899f-db2b7a3139ea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NjM1MzcsIm5iZiI6MTcxOTQ2MzIzNywicGF0aCI6Ii8xNzI2NDU4LzM0MzA4Mjk3Ni1jNzk4MTY4Yi02MmJiLTRkMDItODk5Zi1kYjJiN2EzMTM5ZWEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjdUMDQ0MDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTdlODAyZjVlYTE3YjA4YzE3ODZlNGIwMDYxZjlmM2VkMTE1NmJkYzNiNGFlOWZkOGU0Njg5YjE0ZDljYThhMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.iGEPAaQCFk93bbMbc-l6aouLe22gGQdjHFXVLIC626s)
> 
> 然后手动触发编译成功了，等凌晨的定时看看是不是正常
> 
> Update 2：定时任务也编译成功了

到底怎么搞

oXnMe, 2024-06-27:
> 到底怎么搞

我上面那样改过Makefile之后暂时是正常了

mitsukileung, 2024-07-07:
> > 到底怎么搞
> 
> 我上面那样改过Makefile之后暂时是正常了
> 
> 好像是403的问题修复了，我用旧的tag跑了下也能跑通了😂

按照上面修改后的 makefile 确可以通过编译