coolsnowwolf/packages #845: uugamebooster  6.3.10.tar.gz 404

justlovemaki, 2024-08-12:
**云平台编译获取uugamebooster时404，下载不了资源**

2024-08-11T22:52:33.7796071Z SHELL= flock /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/tmp/.uugamebooster-x86_64-v6.3.10.tar.gz.flock -c '  	/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/scripts/download.pl "/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/dl" "uugamebooster-x86_64-v6.3.10.tar.gz" "17fb275fe666ec3d03ba3b41c2441156" "" "https://uu.gdl.netease.com/uuplugin/openwrt-x86_64/v6.3.10/uu.tar.gz?"    '
2024-08-11T22:52:33.8087813Z + curl -f --connect-timeout 20 --retry 5 --location https://uu.gdl.netease.com/uuplugin/openwrt-x86_64/v6.3.10/uu.tar.gz?/uugamebooster-x86_64-v6.3.10.tar.gz
2024-08-11T22:52:33.8154058Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2024-08-11T22:52:33.8155346Z                                  Dload  Upload   Total   Spent    Left  Speed
2024-08-11T22:52:33.8156035Z 
2024-08-11T22:52:34.1108035Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:35.4415518Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:36.4428311Z   0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
2024-08-11T22:52:36.7094811Z   0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
2024-08-11T22:52:36.7097455Z   0   829    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
2024-08-11T22:52:36.7099391Z curl: (22) The requested URL returned error: 403
2024-08-11T22:52:36.7127730Z Download failed.
2024-08-11T22:52:36.7130977Z + curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/uugamebooster-x86_64-v6.3.10.tar.gz
2024-08-11T22:52:36.7205443Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2024-08-11T22:52:36.7207440Z                                  Dload  Upload   Total   Spent    Left  Speed
2024-08-11T22:52:36.7208103Z 
2024-08-11T22:52:36.8858947Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:36.8860733Z   0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:36.8862056Z curl: (22) The requested URL returned error: 404
2024-08-11T22:52:36.8887473Z Download failed.
2024-08-11T22:52:36.8890109Z + curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/uugamebooster-x86_64-v6.3.10.tar.gz
2024-08-11T22:52:36.8965173Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2024-08-11T22:52:36.8967328Z                                  Dload  Upload   Total   Spent    Left  Speed
2024-08-11T22:52:36.8968243Z 
2024-08-11T22:52:37.0977040Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:37.2938050Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:37.2939855Z   0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:37.2954657Z curl: (22) The requested URL returned error: 404
2024-08-11T22:52:37.2982958Z Download failed.
2024-08-11T22:52:37.2986391Z + curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/uugamebooster-x86_64-v6.3.10.tar.gz
2024-08-11T22:52:37.3079309Z   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
2024-08-11T22:52:37.3080985Z                                  Dload  Upload   Total   Spent    Left  Speed
2024-08-11T22:52:37.3083701Z 
2024-08-11T22:52:37.6361779Z   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:37.6363428Z   0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
2024-08-11T22:52:37.6372201Z curl: (22) The requested URL returned error: 404
2024-08-11T22:52:37.6394665Z Download failed.
2024-08-11T22:52:37.6395898Z No more mirrors to try - giving up.
2024-08-11T22:52:37.6411070Z make[3]: *** [Makefile:86: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/dl/uugamebooster-x86_64-v6.3.10.tar.gz] Error 2
2024-08-11T22:52:37.6414103Z make[3]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/feeds/packages/net/uugamebooster'
2024-08-11T22:52:37.6416651Z time: package/feeds/packages/uugamebooster/compile#0.30#0.11#3.99
2024-08-11T22:52:37.6425022Z     ERROR: package/feeds/packages/uugamebooster failed to build.
2024-08-11T22:52:37.6430142Z make[2]: *** [package/Makefile:116: package/feeds/packages/uugamebooster/compile] Error 1
2024-08-11T22:52:37.6437209Z make[2]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
2024-08-11T22:52:37.6446121Z make[1]: *** [package/Makefile:110: /home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-08-11T22:52:37.6451796Z make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt'
2024-08-11T22:52:37.6465354Z make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/owrt/include/toplevel.mk:231: world] Error 2
2024-08-11T22:52:37.6497647Z ##[error]Process completed with exit code 2.

mitsukileung, 2024-08-14:
https://github.com/coolsnowwolf/packages/issues/828

justlovemaki, 2024-08-14:
> #828

他是403，我这个报错是404啊，错误码都不一样

mitsukileung, 2024-08-14:
> > #828
> 
> 他是403，我这个报错是404啊，错误码都不一样

你不看下面的回复的吗？不是有解决办法吗？

justlovemaki, 2024-08-14:
> > > #828
> > 
> > 
> > 他是403，我这个报错是404啊，错误码都不一样
> 
> 你不看下面的回复的吗？不是有解决办法吗？

你不看我提供的日志么？第一次403，后面是404. 所以你是觉得uu官方403或其它异常情况时，切换到其它源去下载，结果返回404不是一个需要解决的问题咯？

justlovemaki, 2024-08-14:
[#d3a54ad](https://github.com/coolsnowwolf/packages/commit/d3a54ad2380646524201cdfae7bb53efa4f0173f)

justlovemaki, 2024-08-14:
[#d3a54ad](https://github.com/coolsnowwolf/packages/commit/d3a54ad2380646524201cdfae7bb53efa4f0173f)
@Sunshine-Planet 能再更新下这个文件么？[net/uugamebooster/Makefile](https://github.com/coolsnowwolf/packages/commit/d3a54ad2380646524201cdfae7bb53efa4f0173f#diff-abaa968d0704cc4c5d8c14900d05a4087d5ffa791f1bdd91463f259204470af4 "net/uugamebooster/Makefile")

justlovemaki, 2024-08-17:
Update Feeds流程最后添加如下命令：

justlovemaki, 2024-08-17:
Update Feeds流程最后添加如下命令：

```
sed -i "s/openwrt-\$(UU_ARCH)\/\$(PKG_VERSION)\/uu.tar.gz?/openwrt-\$(UU_ARCH)\/\$(PKG_VERSION)\//g" feeds/packages/net/uugamebooster/Makefile
sed -i "s/\$(PKG_NAME)-\$(UU_ARCH)-\$(PKG_VERSION).tar.gz/uu.tar.gz/g" feeds/packages/net/uugamebooster/Makefile
```

参考[#828](https://github.com/coolsnowwolf/packages/issues/828)，可解决uugamebooster升级版本后，下包失败的问题。

closed 2024-08-17 by justlovemaki