coolsnowwolf/packages #866: 请大佬修复下SoftEther VPN

psonys, 2024-11-01:
Please make sure that the issue subject starts with `<package-name>: `SoftEther VPN

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```
[image]

wangtieying, 2024-11-04:
softether vpn不需要luci控制，直接安装就好了，不用安装luci

youring, 2024-11-11:
> softether vpn不需要luci控制，直接安装就好了，不用安装luci

实际效果可能是这样，但少了UI，也就少了启用开关....不方便重启SoftetherVPN服务端
[image]

wangtieying, 2024-11-11:
> > softether vpn不需要luci控制，直接安装就好了，不用安装luci
> 
> 实际效果可能是这样，但少了UI，也就少了启用开关....不方便重启SoftetherVPN服务端
> [image]
> 

在那个系统的启动项里面能重新启动，

youring, 2024-11-11:
>在那个系统的启动项里面能重新启动，

感谢~  //从没用过这地方....
[image]
简单记录下，原来的界面消失后，如何管理SoftetherVPN服务器：
系统--启动项，搜索softether，定位到softethervpnserver进程管理，其中`已启用/已禁用`标识是否随OpenWrt启动，后面三个按钮`启动`、`重启`、`停止`点击后不会有UI提示，需要自行用Softether VPN Server管理器连接确认是否`启动/重启/停止`成功

如果启动项这个地方能有提示就好了。

另求推荐，有没有比OpenWrt/LEDE自带的进程管理更好用的luci app，方便进程启停管理？

wangtieying, 2024-11-11:
> 感谢~  //从没用过这地方....
> [image]
> 简单记录下，原来的界面消失后，如何管理SoftetherVPN服务器：
> 系统--启动项，搜索softether，定位到softethervpnserver进程管理，其中`已启用/已禁用`标识是否随OpenWrt启动，后面三个按钮`启动`、`重启`、`停止`点击后不会有UI提示，需要自行用Softether VPN [...]

他这个有就不错了，我这推了个softether的升级好几天了没人管，能用就行了总比手动命令强，softether这玩意感觉在openwrt下有问题，版本对不上配置文件不自动保存？，从启openwrt或者直接断电的话配置文件不保存，得手动从启软件才保存😥

coolsnowwolf, 2026-09-05:
该 issue 的讨论已确认：SoftEther VPN Server 不依赖单独的 LuCI 页面，可在“系统 → 启动项”中对 `softethervpnserver` 执行启用、启动、重启和停止；提问者也已确认找到该管理入口。当前没有可复现的 packages 包缺陷，因此关闭。若有服务端进程或配置保存问题，请基于当前 `softethervpn5` 版本附日志另开 issue。

closed 2026-09-05 by coolsnowwolf